teslamotors/fleet-telemetry #266: What exactly is a Signal?

Urkman, 2024-12-12:
What is a signal and how they are counted?

And why do my Tesla send signal like this?
[image]

Or all these changes while parked?
[image]

All these problems will lead to why higher payments then necessary :(

Urkman, 2024-12-12:
Here is the missing image...
[image]

Bre77, 2024-12-12:
You have configured the vehicle to send you that data with `interval_seconds: 10`. If  you dont want to pay to get the data that often, then increase the interval. Unfortuantely floating point values like these change a lot.

I do agree that maybe sending gpsHeading and location when the vehicle is in park could be a little redundant, but hey, maybe its on the back of a tow truck?

Urkman, 2024-12-12:
`Unfortuantely floating point values like these change a lot.... Unfortuantely floating point values like these change a lot.`
Sorry, they should not change back and forth and it should not send the same value again and again...

For other values like odometer this is not happening. For odometer I get exactly one signal when it is parked...

jbanyer, 2024-12-12:
I would agree that it would make sense to have some sensible throttling on floating point value changes. Surely nobody cares about a 0.0001% change in battery SoC or energy remaining?

Urkman, 2024-12-12:
And one more from my last drive.
These are 3 signals in one second, why are they not grouped?

[image]

bassmaster187, 2024-12-12:
> And one more from my last drive. These are 3 signals in one second, why are they not grouped?

Even if they are grouped, they count as 3 signals.

bassmaster187, 2024-12-12:
I count every single signal and that matches to the billing.

bassmaster187, 2024-12-12:
Sorry, I am wrong. I was on par on 07th Dec., but it seems like today I'm far away from Tesla. Tesla should really make it clear. 

[image]

[image]

dirkbonhomme, 2024-12-12:
Where can you track the number of signals and Tesla billing?

Bre77, 2024-12-12:
> Where can you track the number of signals and Tesla billing?

The Fleet Telemetry server has a Statd and Prometheus metric output which you can use to count signals received. Your Tesla dashboard in billable countries shows your current billed volumes.

bassmaster187, 2024-12-12:
> > Where can you track the number of signals and Tesla billing?
> 
> The Fleet Telemetry server has a Statd and Prometheus metric output which you can use to count signals received. Your Tesla dashboard in billable countries shows your current billed volumes.

I think I should update to the newest version. My istallation is from Feb. 2024 - never change an running system :-) 

Just override all files and restart the docker?

[image]

Urkman, 2024-12-13:
> Tesla should really make it clear.

Yes, that is a huge problem. 
No support and no documentation :( But the billing is fine...

patrickdemers6, 2025-01-22:
A signal is one field's value delivered to your server. 3 fields in one message counts as 3 signals.

Please submit a support ticket if you still don't think we're tracking usage properly @bassmaster187.

> I would agree that it would make sense to have some sensible throttling on floating point value changes. Surely nobody cares about a 0.0001% change in battery SoC or energy remaining?

minimum_delta is now available and provides exactly this.

closed 2025-01-22 by patrickdemers6