teslamotors/fleet-telemetry #267: block / restrict Endpoint vehicle_data

bassmaster187, 2024-12-12:
My App doens't need vehicle_data anymore. Unfortunately there are hundreds of old apps still using it and I'll be charged. If we will be charged for data commands, we need to be able to control it. e.g. block a VIN or block it at all or limit it to a number of calls a day. For commands it is very eazy: we have a proxy and block all access to set_charging_amps and wake_up. But to data requests everything is out of our scope. 
Another solutions would be to restrict data calls to a whitelist of IPs. So we need to forward all trafic and make sure the traffic is filtered on our side.

Urkman, 2024-12-12:
That is also a good point.

bassmaster187, 2024-12-12:
I was very happy to see that 3 APPs were updated to the latest version yesterday. Unfortunately, someone installed an old version that he either found somewhere on the internet or from a backup...

If we pay for anything, we should have the control to it.

ThomasAlxDmy, 2025-01-16:
That's a good feedback - the only way to do that at the moment is to send everything through a proxy and prevent vehicle_data.