teslamotors/fleet-telemetry #270: Error on telemetry docker restart

Urkman, 2024-12-13:
Today I pulled the last repo changes on my server and restarted and recreated the docker container.
After this I only get the attached errors.

> panic: open /config/0013_chain.pem: permission denied
the file "0013_chain.pem" is there and also has permissions for all to read.

Anyone with an idea on this?

Log entry:

> goroutine 1 [running]:
> app-1  | main.main()
> app-1  | 	/go/src/fleet-telemetry/cmd/main.go:50 +0x106
> app-1  | 2024/12/13 12:28:22 maxprocs: Leaving GOMAXPROCS=2: CPU quota undefined
> app-1  | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"config_skipping_empty_metrics_provider","time":"2024-12-13T12:28:22Z"}
> app-1  | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"starting_server","time":"2024-12-13T12:28:22Z"}
> app-1  | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"kafka_registered","namespace":"telemetry","time":"2024-12-13T12:28:22Z"}
> app-1  | {"activity":true,"context":"fleet-telemetry","dispatcher":"logger","level":"info","msg":"attempting_to_close","time":"2024-12-13T12:28:22Z"}
> app-1  | {"activity":true,"context":"fleet-telemetry","dispatcher":"kafka","level":"info","msg":"attempting_to_close","time":"2024-12-13T12:28:22Z"}
> app-1  | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"stopped_server","time":"2024-12-13T12:28:22Z"}
> app-1  | panic: open /config/0013_chain.pem: permission denied
> app-1  |
> app-1  | goroutine 1 [running]:
> app-1  | main.main()
> app-1  | 	/go/src/fleet-telemetry/cmd/main.go:50 +0x106

Urkman, 2024-12-13:
I had to switch back to the commit "03528fd" to get it back working again...
What can cause this problem? Now everything is working fine again...

EmilOlovsson, 2024-12-13:
Check file permissions on file: /config/0013_chain.pem:

Fleet-telemetry is using this image:
FROM gcr.io/distroless/cc-debian11:nonroot

I quick fix is to run the docker container fleet-telemetry as user "root".
[https://stackoverflow.com/a/48730146](https://stackoverflow.com/a/48730146)

Urkman, 2024-12-14:
> I quick fix is to run the docker container fleet-telemetry as user "root".
> https://stackoverflow.com/a/48730146

This seems to be the trick...

Thanks :)

closed 2025-01-22 by patrickdemers6