teslamotors/fleet-telemetry #271: Too many updates due to float change

rawmean, 2024-12-14:
Many of the fields that telemetry streams are floating points and even when we set the interval to seconds or minutes, they do update. Theis did not matter much before but now that we are going to pay for these updates, can you please fix it so they don’t update each time a value at the 10th decimal digit changes?  Here is an example:

[image]

Bre77, 2024-12-14:
I also have this issue, and as you can see with Inside Temp, the actual usable value only has 1 decimal place, the rest is floating point error. I don't think we shouldnt be charged for changes in floating point errors.

bassmaster187, 2024-12-14:
They will be charged / sent, even if they didn't change. At least for PackCurrent

[image]

Urkman, 2024-12-14:
We already had this in another thread... 
Some values are send only on change, some values just change their values back and forth and some other values are send over and again without any change...

The whole system is totally unreliable and simply not production ready :(

ThomasAlxDmy, 2025-01-16:
@Urkman Not sure how these type of statement help your case. Feel free to hit me up on X to share your opinions and we can debate there what makes a system production ready or not ;). 

@rawmean There's a config called minium_delta that you can pass (the same way you'd configure a field interval) that was released in the early December software update. Details there: https://developer.tesla.com/docs/fleet-api/announcements#2025-01-09-new-fleet-telemetry-configuration-options

closed 2025-01-16 by ThomasAlxDmy