teslamotors/fleet-telemetry #274: How to un-pair a fleet key?

sjmiller609, 2024-12-17:
Fleet keys are paired with vehicles via the app, after the user goes to a domain like tesla.com/_ak/example.com https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#key-pairing

Looking in Tesla app on iOS, control -> locks, but I don't see any third party virtual keys listed here, but the vehicle is connecting and streaming telemetry data to the server. Maybe I'm not seeing it, does it look exactly like a key fob? Does it say the domain of the key on it?

I looked in account settings on the web, I see a way to configure scopes for oauth but not to unpair a vehicle from the fleet key. Does revoking the third party's access via oauth unpair associated fleet keys?

sjmiller609, 2024-12-17:
Revoking and re-adding via oauth does not seem to remove the fleet key from the vehicle

iainwhyte, 2024-12-17:
Hi @sjmiller609 I wonder if perhaps what threw me is your problem, the list of keys can scroll but no indication it can...
Here is what a Bluetooth and Fleet keys looks like:
[image]

Bre77, 2024-12-17:
Speaking from experience, the **only** way to remove a vitual key is from inside the vehicle with a keycard.

closed 2025-01-03 by sjmiller609