teslamotors/fleet-telemetry #276: Report typo in unsupported interval and some question.

junghoon-vans, 2024-12-18:
There is a typo in the response value if you set an `unsupported interval`. 
Is it a value that is coming directly from the car? It's probably really small. But I'm reporting it because it would be nice to fix it. 

Below is an example of this.

```json
{
    "activity": true,
    "context": "fleet-telemetry",
    "data": [
        {
            "Body": "",
            "CreatedAt": 1734416033,
            "Name": "unsupported_field",
            "Tags": {
                "field_name": "TpmsSoftWarnings",
                "name": "6b0f17e96a64-4d78-beec-a425425aeb8b"
            }
        },
        {
            "Body": "",
            "CreatedAt": 1734416033,
            "Name": "unsuported_interval",  // typo
            "Tags": {
                "authorized": "60000",
                "field_name": "ExteriorColor",
                "name": "6b0f17e96a64-4d78-beec-a425425aeb8b",
                "requested": "10.000000"
            }
        }
    ]
}
```
And one question, is there any charge for these unsupported fields?

agbpatro, 2025-01-03:
Thanks for reporting this. We will get the typo fixed in one of the upcoming releases

agbpatro, 2025-01-03:
Also to answer `And one question, is there any charge for these unsupported fields?`
it should not

closed 2025-01-15 by junghoon-vans