teslamotors/fleet-telemetry #277: Return all Unknown Field(s) on first request to endpoint fleet_telemetry_config create

andrewdevelopz, 2024-12-18:
If I submit unknown Fields to https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-endpoints#fleet-telemetry-config-create

```
...
        "fields": {
            "ClimateKeeperModeState": {
                "interval_seconds": 60
            },
            "Experimental_1": {
                "interval_seconds": 60
            },
            "Experimental_2": {
                "interval_seconds": 60
            }
        },
...
```

It returns 1 bad Field at a time for each request

e.g. 400 response:
```
{
    "response": null,
    "error": "Unknown field ClimateKeeperModeState",
    "error_description": "",
    "txid": <txid>
}
```

Would it be possible to change this so that it returns all the bad Fields on the first request?

So something like this:
```
{
    "response": ["ClimateKeeperModeState", "Experimental_1", "Experimental_2"],
    "error": "Unknown fields",
    "error_description": "",
    "txid": <txid>
}
```

Thank you.

ThomasAlxDmy, 2025-01-08:
Sounds like a good improvement - we'll add it to the backlog. Thank you!!

andrewdevelopz, 2025-01-08:
@ThomasAlxDmy Thank you, I really appreciate the help.