teslamotors/fleet-telemetry #280: Lots of UNRECOGNIZED on new vehicles...

Urkman, 2024-12-26:
For vehicles with 2024.44.32 (perhaps others too) I get a lot of UNRECOGNIZED currently:

Any idea?

>UNRECOGNIZED(228): ,UNRECOGNIZED(183): ,locked: true,fpWindow: Closed,timeToFullCharge: ,estBatteryRange: 208.83266775082396,ratedRange: 230.56603258229651,detailedChargeState: ,UNRECOGNIZED(195): ,seatHeaterRight: 0,UNRECOGNIZED(204): ,vehicleSpeed: ,rpWindow: Closed,fastChargerPresent: ,UNRECOGNIZED(203): ,seatHeaterRearCenter: 0,UNRECOGNIZED(200): ,UNRECOGNIZED(193): ,sentryMode: Off,seatHeaterRearRight: 0,insideTemp: 9.300000734627247,version: 2024.45.32 26bf7521c499,UNRECOGNIZED(213): ,chargeLimitSoc: 81,doorState: ,acchargingPower: ,dcchargingPower: 0,currentLimitMph: 85,UNRECOGNIZED(220):  ,gear: ,energyRemaining: 62.23999860882759,UNRECOGNIZED(196): ,autoSeatClimateLeft: true,seatHeaterRearLeft: 0,UNRECOGNIZED(191): ,odometer: 23699.926064046005,soc: 80.9271523178808,UNRECOGNIZED(180): ,autoSeatClimateRight: true,fdWindow: Closed,batteryHeaterOn: ,seatHeaterLeft: 0,outsideTemp: 7,location: ,rdWindow: Closed,UNRECOGNIZED(181): ,UNRECOGNIZED(188): ,UNRECOGNIZED(197): ,speedLimitMode: false,UNRECOGNIZED(187):

Bre77, 2024-12-26:
Are you running the latest version of the Fleet Telemetry server? The protobufs were updated for the Holiday update.

Urkman, 2024-12-30:
Yes, I didn't see this update... 
Closed...

closed 2024-12-30 by Urkman