teslamotors/fleet-telemetry #282: Missing fields in telemetry: `charge_miles_added_rated`, `charge_miles_added_ideal`

rawmean, 2024-12-28:
Can you please add these fields to telemetry to make the information closer to once provided by `vehicle_data` endpoint?

ThomasAlxDmy, 2025-01-16:
Can you look at the last definition and tell us what's currently missing from your perspective? Curious to also understand the use-cases

rawmean, 2025-01-16:
> Can you look at the last definition and tell us what's currently missing from your perspective? Curious to also understand the use-cases

I examined the fields in the `vehicle_data.proto` and do not see the equivalent of `charge_miles_added_rated` and 'charge_miles_added_ideal` which exist in the `vehicle_data` endpoint. These two fields enabled us to compute information related to each charging session. 

Since the endpoint for charging history is removed (there are  charging_history endpoints  currently available but one only returns supercharging info and the other one is related to the wall charger) these fields are helpful in creating charging session information.

Of course, one can add state to the dispatch code and record these values for each vin, but it would be great if the these values could be available in the fleet telemetry for reasons mentioned above.  

Thanks

ZayanKhan-12, 2026-09-16:
I went looking at whether this could be contributed as a PR, and it can't — but there may be a usable workaround in the meantime, so sharing both.

**On the request itself:** confirming @rawmean's reading. There is still no equivalent of `charge_miles_added_rated` or `charge_miles_added_ideal` in the `Field` enum as of `main`. Since field numbers are allocated alongside firmware releases (the proto records this, e.g. `// fields 260-269 are first available in firmware version 2026.32`), this genuinely can't be added from outside — which is presumably why @patrickdemers6 labelled it `firmware`. Adding an enum entry locally wouldn't make vehicles emit it, and picking an unused number would actively backfire: 260 would have seemed free when this issue was opened, and it became `GpsAccuracyMeters` in #546.

**A possible interim workaround.** The two `charge_miles_added_*` values are session deltas, and the underlying range signals *are* streamed today, so a charging session can be reconstructed consumer-side without keeping per-VIN state indefinitely — you only need state for the duration of a session:

- `DetailedChargeState` (179) brackets the session. `DetailedChargeStateStarting` / `DetailedChargeStateCharging` open it; `DetailedChargeStateComplete`, `DetailedChargeStateStopped` or `DetailedChargeStateDisconnected` close it.
- `RatedRange` (32) sampled at both ends gives the rated-range delta, which is what `charge_miles_added_rated` reports.
- `IdealBatteryRange` (41) gives the ideal equivalent.
- `ACChargingEnergyIn` (36) / `DCChargingEnergyIn` (34) give energy added for the same window, which is often the more robust number to key billing or reporting off.

Two caveats worth knowing before relying on this:

1. **Units.** The proto does not document units for the range fields. `SettingChargeUnit` (236) reports the vehicle's configured charge unit, so it's worth capturing alongside the range values rather than assuming miles — particularly for a mixed-region fleet.
2. **It's an approximation, not the same number.** A range delta can move during a session for reasons other than energy added (cabin conditioning, thermal management, recalibration), so it won't always tie out exactly with what `vehicle_data` reports. If you need the two to agree precisely, this workaround won't get you there and the firmware-side field is the real answer.

@rawmean — you'd already noted you could hold state in the dispatch code, so this may be exactly what you'd worked out. Posting the specific field numbers in case it saves someone else the lookup, and because the `DetailedChargeState` bracketing makes the state you have to hold quite small.

Separately, I've opened #551 to document the firmware-gating constraint in the README, since this is being asked across quite a few issues (#249, #311, #312, #343, #376, #445, #446, #447) and there's currently nothing in the docs explaining it. That's a docs change only and doesn't affect this request.

rawmean, 2026-09-20:
Thanks @ZayanKhan-12. Yes, I ended up maintaining bunch of state similar to what's described above. It's a firmware issue because the car needs to publish these fields that existed prior to the telemetry model.