teslamotors/fleet-telemetry #283: Car Stops streaming while charging

rawmean, 2024-12-29:
when the car goes to sleep while charging, it stops streaming data. Even when the car is awakened, it won’t resume streaming data until a command (eg, turning on climate) is sent to the car. 

This creates a problem because the data in the app becomes out of sync with the car (the car has been charging while streaming stopped).

Bre77, 2024-12-29:
Your vehicle shouldn't be going to sleep while charging, do you mean while plugged in?

How are you waking your vehicle in your test?

My Model 3 never sleeps while charging and sends Fleet Telemetry as soon as it wakes up.

rawmean, 2024-12-30:
> Your vehicle shouldn't be going to sleep while charging, do you mean while plugged in?
> 
> How are you waking your vehicle in your test?
> 
> My Model 3 never sleeps while charging and sends Fleet Telemetry as soon as it wakes up.

Exactly, I wasn't expecting the car to go to sleep while charging either. It probably didn't but streaming was stopped for some reason (it was actively charging; not just plugged in).

I woke the car by enabling climate using Tesla app.

closed 2025-01-11 by rawmean