teslamotors/fleet-telemetry #284: 412 status when calling fleet_telemetry_config create

csailly, 2024-12-29:
Hey dev team,
I followed the steps from https://dev.to/patrickdemers6/fleet-telemetry-streaming-data-from-your-tesla-2i97
And I have a trouble creating the fleet telemetry config:

Here the request  
```
> POST /api/1/vehicles/fleet_telemetry_config HTTP/2
> Host: fleet-api.prd.eu.vn.cloud.tesla.com
> content-type: application/json
> authorization: Bearer eyJhbGxxxxx
{
  "vins": ["xxxxxxxx"],
  "config": {
    "hostname": "tm3.xxxx.fr",
    "ca": "-----BEGIN CERTIFICATE-----\nxxxx\n-----END CERTIFICATE-----\n-----BEGIN CERTIFICATE-----\nyyyy\n-----END CERTIFICATE-----",
    "fields": {
      "Soc": {
        "interval_seconds": 30
      }
    }
  }
}

```

And here the response with a 412 Precondition failed status

```
HTTP/2 412 
x-xss-protection: 1; mode=block
x-txid: d4a95f9d821d6f78f73143b2f078cdad
content-type: application/json; charset=utf-8
vary: Accept,Accept-Encoding
x-request-id: 249b0bf6-4239-4709-a5a2-a12377fa8633
content-length: 205
x-envoy-upstream-service-time: 8
x-frame-options: DENY
x-content-type-options: nosniff
strict-transport-security: max-age=31536000; includeSubDomains
cache-control: no-cache, no-store, private, s-max-age=0
date: Sun, 29 Dec 2024 21:01:26 GMT
server: envoy
{
  "response": null,
  "error": "Refer to https://developer.tesla.com/docs/fleet-api/fleet-telemetry for instructions setting up fleet-telemetry.",
  "error_description": "",
  "txid": "d4a95f9d821d6f78f73143b2f078cdad"
}
```

The fleet_status seems ok :
```
{
  "response": {
    "key_paired_vins": ["xxxxxxxx"],
    "unpaired_vins": [],
    "vehicle_info": {
      "xxxxxxxx": {
        "firmware_version": "2024.44.25",
        "vehicle_command_protocol_required": true
      }
    }
  }
}
```

Can you tell me what I'm doing wrong please ?

Bre77, 2024-12-29:
The process changed since Patrick's guide. If you havent gone through the CSR process with Tesla Fleet API Support (which they no longer do) you need to send the configure command through the Command Proxy.

https://developer.tesla.com/docs/fleet-api/announcements#2024-08-15-self-service-fleet-telemetry

https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-endpoints#fleet-telemetry-config-create

csailly, 2024-12-30:
Thanks @Bre77 I will use the Command Proxy 👍🏻

csailly, 2024-12-30:
I close the issue  because I was not use the Command Proxy

closed 2024-12-30 by csailly

Rockster160, 2024-12-31:
@Bre77 - I apologize if this is the wrong place, but figured this was better than creating a duplicate post.

I'm having the same issue- everything is nearly identical, so it feels redundant to display it all again, however, using the proxy I still get the same error:
```
POST http://127.0.0.1:3142/api/1/vehicles/fleet_telemetry_config → https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/fleet_telemetry_config
========================  Payload  =========================
{
  vins: ["xxxxx"], 
  config: {
    alert_types: ["service"], 
    fields: {ChargeState: {interval_seconds: 60}, VehicleSpeed: {interval_seconds: 60}, ...}, 
    ca: "-----BEGIN CERTIFICATE-----\nxxxxxxxx\n-----END CERTIFICATE-----\n", 
    hostname: "https://mydomain.com/tesla"
  }
}
========================  Headers  =========================
{
  "CONTENT-TYPE": "application/json", 
  VERSION: "HTTP/1.1", 
  ACCEPT: "*/*", 
  "USER-AGENT": "xxxxxx", 
  AUTHORIZATION: "Bearer xxxxxxxxxx", 
  "ACCEPT-ENCODING": "gzip;q=1.0,deflate;q=0.6,identity;q=0.3", 
  HOST: "127.0.0.1:3142"
}
========================  Response =========================
412 
{
  response: null,
  error: "Refer to https://developer.tesla.com/docs/fleet-api/fleet-telemetry for instructions setting up fleet-telemetry.",
  error_description: "",
  txid: "4cce3185b90985627c6033dc5c3b3b46"
}
```

These ARE running through the proxy, and I'm able to send commands directly to the vehicle using the proxy, but I'm still getting the 412 when trying to create the `fleet_telemetry_config`

Like the other dev showed, fleet_status is good:
```
POST http://127.0.0.1:3142/api/1/vehicles/fleet_status → https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/fleet_status
========================  Payload  =========================
{ vins: ["xxxxx"] }
========================  Headers  =========================
{
  "CONTENT-TYPE": "application/json", 
  VERSION: "HTTP/1.1", 
  ACCEPT: "*/*", 
  "USER-AGENT": "xxxxxx", 
  AUTHORIZATION: "Bearer xxxxxxxxxx", 
  "ACCEPT-ENCODING": "gzip;q=1.0,deflate;q=0.6,identity;q=0.3", 
  HOST: "127.0.0.1:3142"
}
========================  Response =========================
{
  response: {
    key_paired_vins: ["xxxxx"], 
    unpaired_vins: [], 
    vehicle_info: {
      "xxxxx": {
        firmware_version: "2024.45.32", 
        vehicle_command_protocol_required: true
      }
    }
  }
}
```

The OAuth Client Details are set up like so:
```
Allowed Origin(s)
https://mydomain.com

Allowed Redirect URI(s)
https://mydomain.com/webhooks/auth

Allowed Returned URL(s)
https://mydomain.com
```

I wasn't sure if using a path instead of a subdomain for the telemetry callbacks was allowed or not.

I'd love to get this working since we're currently using the polling strategy against the `vehicle_data` endpoint which is not ideal. I'm happy to provide any more details that would help get this figured out!

Bre77, 2024-12-31:
> I wasn't sure if using a path instead of a subdomain for the telemetry callbacks was allowed or not.

The path is the only thing that looked wrong to me. How have you implemented that? Because you can't have a reverse proxy in front of the Fleet Telemetry server, and I don't recall it supporting paths.

I wouldn't have expected this to make the config endpoint fail but Tesla possibly has checks in place for things like this, but not useful errors.

Rockster160, 2024-12-31:
Actually, that raises a pretty obvious issue that I'd largely forgotten about. **Tesla will be sending the packets via the _command proxy server_.** That piece didn't completely click for me.

tl;dr: I am using a double proxy to run the command proxy server locally (home server), using a tunnel to access it from my main server. Because Tesla will be sending the commands through the proxy server... Things get a little hairy.

-----

I've had a set up like this for a few years (long before it was officially supported. 😅)  and at one point maybe 2 or so years ago it seemed as though Tesla marked my domain as spam or something and blocked all auth requests from it. Oddly once I'd generated an `access_token`, I was able to make any/all requests from the domain, but in order to get the access_token, I had to pass the refresh_token to a private/home server, which would refresh the access_token and then pass it back to the real/live/public server. So essentially I had to use a private proxy server for the auth commands. Then when the Telemetry/Command Proxy stuff came about, since I'd have to pass all of the auth details to the local proxy server anyway, 

I can probably set up a tunnel on the server that uses my home proxy. Not a huge fan of opening a `*` tunnel to my home server, but it might be needed unless I can contact Tesla dev support and have them unblock my domain.

Bre77, 2024-12-31:
> tl;dr: I am using a double proxy to run the command proxy server locally (home server),

That's fine as long as things are being properly rewritten.

But your config has the hostname of the Fleet Telemetry server with a path, and you can't receive the mTLS websocket connections through a reverse proxy. You can through a network (layer 4) load balancer, but anything layer 7 will break the mTLS, and the server is expecting websocket connections on /