teslamotors/fleet-telemetry #286: My Teslas stopped sending signals

Urkman, 2024-12-30:
Hi,

my Tesla stopped sending signals this morning :( 
Other Teslas using my App are running fine. 

What could cause this problem and what is the best way to debug something like this?
And prevent this in the future? 

Thanks,
Urkman

Urkman, 2024-12-30:
Aha.. He is back... A Reset from the Tesla did the trick... But why?

jlestel, 2025-01-08:
I see the same thing. The get on fleet_telemetry_config returns: `{"response":{"synced":true,"config":null}}`

Was it a Hard Reboot that allowed you to solve this?

ThomasAlxDmy, 2025-01-16:
for these type of specific issue we might need precise information to debug. If you run into this more please open a JIRA request or email support [fleetapisupport@tesla.com](mailto:fleetapisupport@tesla.com)

closed 2025-01-16 by ThomasAlxDmy