teslamotors/fleet-telemetry #287: What is the difference between EnergyIn fields?

marklindsey, 2025-01-02:
Regarding the two fields ACChargingEnergyIn and DCChargingEnergyIn, how do these fields differ? For my use case, I'd like to capture when energy is added irrespective of if it is from a 1 kW, 6kW, or 250 kW charger. My first guess would be that AC refers to Level 1 and Level 2 chargers whilst DC refers to Level 3 chargers. However, when I stream data from my Level 2 (6 kW) charger, I receive both AC and DC energy in values that differ which I wouldn't expect. Any insight would be appreciated. Thank you.

Here's an example output I'm receiving:
`"ACChargingEnergyIn":"16.179500241092526",
"ACChargingPower":"5.900000087916851",
"CreatedAt":"2024-12-30T05:15:34Z",
"DCChargingEnergyIn":"15.359999656677246",
"RatedRange":"148.76711996246692"`

rawmean, 2025-01-04:
I think `DCChargingEnergyIn` is the energy that has actually been added to the battery. This matches what Tesla app shows.
I think `ACChargingEnergyIn` is the energy that is given to the car. This is more than `DCChargingEnergyIn` and the difference is due to charging efficiency. 
Since there is no documentation, I may be wrong about `ACChargingEnergyIn`

marklindsey, 2025-01-04:
Thanks for the context @rawmean. I hadn't had a chance to confirm that `DCChargingEnergyIn` matched `charge_state.charge_energy_added`.

closed 2025-01-04 by marklindsey