teslamotors/fleet-telemetry #288: ShiftState value inconsistent with proto

morganofslo, 2025-01-02:
We are attempting to read ShiftState, however the values we receive don't seem to conform to the[ defined enum](https://github.com/teslamotors/fleet-telemetry/blob/c3daf762faee7905946a4b4da27fa4ff6034765a/protos/vehicle_data.proto#L203) values, which is concerning.  For example, instead of 'ShiftStateD', we receive 'D'.  This doesn't seem consistent with other enum behaviors; for example, for DetailedChargeState, we receive 'DetailedChargeStateDisconnected' not 'Disconnected'

agbpatro, 2025-01-02:
hmm which field are you reading? Value of [`Gear`](https://github.com/teslamotors/fleet-telemetry/blob/c3daf762faee7905946a4b4da27fa4ff6034765a/protos/vehicle_data.proto#L20) should be ShiftState enum

Bre77, 2025-01-02:
If you add `"prefer_typed":true` to your vehicle config you get the full value from the ENUM. Without that you only get the bit at the end that changes for all ENUMs.

rileymd88, 2025-01-06:
Just adding that I see the same behaviour. Would be nice to have examples/enums for all fields so we know what to expect

Bre77, 2025-01-07:
You can get most of the way there by looking through https://github.com/Teslemetry/tesla-fleet-telemetry/blob/main/protos/vehicle_data.proto

But thats mostly for newer fields rather than the originals.

closed 2025-01-16 by patrickdemers6