teslamotors/fleet-telemetry #289: googlepubsub.Producer.produce() checks for topic existence every time, consumes quota

chriselion, 2025-01-03:
`googlepubsub.Producer.produce()` checks whether the topic exists (and creates it if it doesn't) before publishing every message:

https://github.com/teslamotors/fleet-telemetry/blob/4034cea43e3c80faf3b94c69039472a1777cb3d0/datastore/googlepubsub/publisher.go#L88-L100

Unfortunately, each check consumes part of a quota; at a high volume of messages, they will start failing with errors like
```
rpc error: code = ResourceExhausteddesc = Quota exceeded for quota metric 'Administrator operations' and limit 'Administrator operations per minute' of service 'pubsub.googleapis.com' for consumer 'project_number:1234567890'.
error details: name = ErrorInfo reason = RATE_LIMIT_EXCEEDED domain = googleapis.com metadata = map[consumer:projects/1234567890 quota_limit:administratorPerMinutePerProject quota_limit_value:6000 quota_location:global quota_metric:pubsub.googleapis.com/administrator service:pubsub.googleapis.com]
```

And more unfortunately, this is a system limit and can't be extended.

Some possible solutions I can think of:
* Cache the existence check, either indefinitely, or for a few minutes
* Only check if publishing fails (aka "better to ask forgiveness than permission")
* Add an option to the pubsub configuration to allow toggling the behavior

agbpatro, 2025-01-07:
fixed in https://github.com/teslamotors/fleet-telemetry/releases/tag/v0.5.0

chriselion, 2025-01-07:
Thanks! We tried it out this morning and seems to be working well.