teslamotors/fleet-telemetry #293: unsupported_firmware issue

Daeell, 2025-01-08:
hello

I have configured fleet-telemetry and vehicle-command proxy and also registered a Tesla fleet key.

so If I send a request with the following payload, 

```
    "vins": [
        "my_car_vin"
    ],
    "config": {
        "hostname": "telemetry.domain.com",
        "port": 443,
        "ca": ca_cert,
        "fields": {
            "Soc": {
                "interval_seconds": 30
            }
        }
```

I get a response like this

```
{"response":{"updated_vehicles":0,"skipped_vehicles":{"missing_key":[],"unsupported_hardware":[],"unsupported_firmware":["my_car_vin"]}}}
```

Tesla's official documentation states that

> VINs may be rejected for a few reasons:
> ...
> unsupported_firmware: Vehicles running firmware version earlier than 2024.26.

So I checked the software for that vehicle and found the following.

```
"software_update": {
        "download_perc": 0,
        "expected_duration_sec": 2700,
        "install_perc": 1,
        "status": "downloading_wifi_wait",
        "version": "2024.32.7"
},
```

So, based on what the official documentation says, this shouldn't be the case. 
Is it possible that I should always have the latest version so that this doesn't happen?

Bre77, 2025-01-08:
What you have posted shows there is a firmware update available, that has not been installed. You need to check the `vehicle_state.car_version` to see the current firmware version.

Daeell, 2025-01-08:
@Bre77 

Ah, thank you. 

"car_version": "2024.14.8 db4e5a400f8d",

I checked the vehicle version and realized that it was unsupported_firmware.

So, when I tried to update the vehicle further, I didn't see any updateable versions.

"software_update": {
          "download_perc": 0,
          "expected_duration_sec": 2700,
          "install_perc": 0,
          "status": "",
          "version": " "
        }
        
I check vehicle_data again and it looks like this, but suddenly there is no updateable firmware?

Bre77, 2025-01-08:
> @Bre77
>
> Ah, thank you.
>
> "car_version": "2024.14.8 db4e5a400f8d",
>
> I checked the vehicle version and realized that it was unsupported_firmware.
>
> So, when I tried to update the vehicle further, I [...]

That's how it should look when there are no updates available.

closed 2025-01-08 by Daeell