teslamotors/fleet-telemetry #295: New Software, new problems :(

Urkman, 2025-01-08:
Today I installed 2024.45.32.1 and now I got strange climate values :(

When I enable or disable the climate (using the Tesla App) I do not get "hvacAcenabled", but I get "hvacSteeringWheelHeatLevel".
When I enable or disable defrost mode, with climate disabled (also using Tesla App) I get changes to "defrostMode".
When I enable defrost mode, with climate enabled, I get the same as when climate is disabled.

And now:
When I disable defrost mode, with climate enabled, I get the first "hvacAcenabled: true" and then "hvacAcenabled: false". But the climate is still enabled :(

So, is there something wrong and do I miss something?

And I get a lot of location changes, but my car is parked :(

[image]

Urkman, 2025-01-08:
And I keep getting the location updates on my parked car.
I checked my logs and this was not the case before 2024.45.32.1

[image]

bassmaster187, 2025-01-08:
That'll be expensive, if Tesla gonna charge us for this..

dirkbonhomme, 2025-01-09:
Is the car plugged in? Or maybe a door hasn't closed properly? I would expect the car to go asleep after ±20min 😞

Urkman, 2025-01-16:
> I would expect the car to go asleep after ±20min

Sentry Mode was enabled...

rawmean, 2025-01-18:
> And I keep getting the location updates on my parked car. I checked my logs and this was not the case before 2024.45.32.1
> 
This is related to floating point bug. Tesla's solution is to add `minimum_delta` to the server config to suppress changes that are smaller than `minimum_delta`. The catch is that this only works if the car has a recent firmware which is currently sent to a small fraction of cars.

I wish Tesla would not start charging at least until the firmware version that supports `minimum_delta` is installed widely. 

@patrickdemers6

chavezoz, 2025-01-19:
@Urkman , @rawmean , any idea how to set _minimum_delta_ to get past the repeated _GPS Location_ duplication issue whereby it's sending the same co-ordinates over and over while the vehicle is stationary? 

The announcement states _minimum_delta_ must be an int thus not sure how that would work.

Adminius, 2025-01-22:
Reduced "precision" can save a lot of traffic/data:
PackVoltage: 0 decimals
PackCurrent: 1 decimals

[image]

patrickdemers6, 2025-01-23:
We are sending too much precision when `prefer_typed` is not set to true. With `prefer_typed` true, the values are sent as floats, meaning no additional data is used.

Urkman, 2025-01-24:
The latest update 2024.45.32.2 fixes the "hvacAcenabled" not being send problem.
So I close the ticket.

closed 2025-01-24 by Urkman