teslamotors/fleet-telemetry #296: What is the actual precision of streaming decimal

marklindsey, 2025-01-08:


ThomasAlxDmy, 2025-01-16:
We recently released minimum_delta configuration to solve for this problem https://developer.tesla.com/docs/fleet-api/announcements#2025-01-09-new-fleet-telemetry-configuration-options

closed 2025-01-16 by ThomasAlxDmy

Urkman, 2025-01-16:
Is there any documentation on this topic or just the announcement?

Bre77, 2025-01-16:
> Is there any documentation on `prefer_typed`? Is this a per field setting too?

It's not per field, it's for the entire config.

Feel free to use this as better documentation than Tesla's: https://api.teslemetry.com/docs/#tag/vehicle/POST/api/1/vehicles/fleet_telemetry_config

Urkman, 2025-01-16:
Yes, but what else will this flag change?

rawmean, 2025-01-16:
I set `prefer_typed` to true and do not notice any change with my car with is on 2024.44.25.2 (which is older than the version required for the `minimum_delta` parameter. But `prefer_typed ` itself did not change the format of the fields. Is that expected?

marklindsey, 2025-01-17:
The root cause of this issue seems to be converting float32 values into float64 values, which is causing erroneous random insignificant rounding digits at the end of the number. To combat this, on a per field basis, I'll plan to keep track of the expected precision, e.g. 0.01, and then round accordingly.

rawmean, 2025-01-17:
> The root cause of this issue seems to be converting float32 values into float64 values, which is causing erroneous random insignificant rounding digits at the end of the number. To combat this, on a [...]

That doesn't prevent the car from sending the data and the associated cost, right? For that the car firmware needs to change (which apparently the more recent ones mentioned in the announcement for `minimum_delta` does)