teslamotors/fleet-telemetry #297: NAT/DNS issue in internal network - how does a car resolve and cache DNS records?

slashrsm, 2025-01-08:
I have a fleet telemetry stack running on a server in my local network. I have to use NAT to forward requests from the internet to the server. 

When the car is away from home everything works great. When I return home and car connects to my WiFi it is unable to connect to the fleet telemetry server. 

DNS records for the telemetry servers differ depending on where you are. When away they will resolve to the public IP of my router and when at home they will resolve to the internal IP of the server. 

Based on the error messages that I get from the car it seems that it fails to connect because it never got the internal DNS record. It is still trying to connect to the public IP address, which at that point is my router and not the fleet telemetry server. 

My questions are:

1. How long will the car cache DNS records? Will it try to re-fetch them when connected to a WiFi?
2. Will car use the DNS server that the DCHP server propagates or it always uses its own?

Anyone else experienced the same problem? How did you resolve it?

patrickdemers6, 2025-01-22:
Specific behavior of the vehicle regarding DNS isn't available.

closed 2025-01-22 by patrickdemers6

rs38, 2025-02-13:
Don't bother with the problems you cannot solve (DNS client settings of the car). :)
you can rule out misconfiguration of you internal DNS config? Do you use a DynDNS service?

slashrsm, 2025-02-13:
I ended up solving this problem with a few routing/NAT rules that ensure packets get to the right machine even if the DNS record isn't correct.

bassmaster187, 2025-02-13:
I think you have a so called "NAT Loopback" or "Hairpinning" Problem. Some home routers can handle this, some not. 

[image]