teslamotors/fleet-telemetry #303: InvalidSignature: unknown issuer

dirkbonhomme, 2025-01-13:
We've started migrating many vehicles to Telemetry today but some cars either:
- Never send data through Telemetry
- Send some data and then stop

This is the error we get from the fleet_telemetry_errors API. Any clues?

```
{
    "created_at": "2025-01-13T19:59:18.55743492Z",
    "error": "error loading config file: InvalidSignature: unknown issuer",
    "error_name": "config_manager_error",
    "name": "e26b29af35ac-42f7-a446-xxxxxxx",
    "txID": "a093e020-12c3-47fd-9392-xxxxxx",
    "vin": "XXX"
}
```

Bre77, 2025-01-13:
Do you send your config through the command proxy or did you do the CSR process?

I wonder if Tesla is invalidating the old CSR method.

dirkbonhomme, 2025-01-13:
This is done through the command proxy. It works fine for most cars though..

dirkbonhomme, 2025-01-14:
I have found the issue. Some configs were pushed through a command proxy that was configured with the incorrect private key. So apparently you are able to successfully push telemetry config to a car with the wrong signature and it's only when sending data to telemetry service (?) that the car checks the key? Can someone from Tesla confirm if this is by design?

1. POST /api/1/vehicles/fleet_telemetry_config
   - use JWT credentials for APP A
   - use command proxy with private key for APP A
   - synced = true
   - data is pushed to fleet telemetry

2. POST /api/1/vehicles/fleet_telemetry_config
   - use JWT credentials for APP A
   - use command proxy with private key for APP B
   - synced = true
   - data is no longer pushed to fleet telemetry
   - unknown issuer error

3. POST /api/1/vehicles/fleet_telemetry_config
   - use JWT credentials for APP A
   - use command proxy with private key for APP A
   - synced = true
   - data is pushed to fleet telemetry again

ThomasAlxDmy, 2025-01-16:
That's correct - we currently do a bunch of validations (like do you have access to the vehicle, is your domain correct etc.) but not is the config signed with a key that's on the car before sending it to the vehicle. 

We could potentially do this seems like a good follow up

dirkbonhomme, 2025-01-16:
A good reason for adding additional checks imo is because right now, it's possible to break a working telemetry config

dwery, 2025-01-26:
+1 on the checks, I've hit the same error. Additionally, if that same priv key had been previously loaded on the vehicle via BLE, telemetry didn't start/work. I had to remove it and re-add via the app.