teslamotors/fleet-telemetry #304: MQTT dispatcher publish error

andrew-kennedy, 2025-01-15:
I have telemetry streaming to the fleet-telemetry server and showing up using the logger dispatcher, so I know that's working. I am also running mosquitto in another docker container and I see connections to it from the telemetry server:

```
1736900224: New connection from <ip>:54758 on port 1883.
1736900224: New client connected from <ip>:54758 as 7sCae6JiRSQQ31BgU12vvZ (p5, c1, k60, u'<username>').
```

However, in the fleet telemetry server, after I see an event in the logs with all the fields I have asked the car to publish, I always see the following event
```
{"context":"fleet-telemetry","error":"context canceled","level":"error","msg":"mqtt_publish_error","time":"2025-01-14T16:28:31-08:00","topic_name":"telemetry_V","txid":"e9e8654f32e541c4b0b623-000000033","vin":"*censored*"}
```

And none of the published signals seem to make it to the mqtt server, despite MQTT not showing any client connection errors. I have been able to use the same username and password to connect to the MQTT broker via other services on other containers, and via home assistant, so I think the broker itself is configured correctly?

This is my config:

```
{
  "host": "0.0.0.0",
  "port": 4343,
  "log_level": "debug",
  "json_log_enable": true,
  "namespace": "telemetry",
  "reliable_ack": false,
  "rate_limit": {
    "enabled": false,
    "message_limit": 100
  },
  "records": {
    "alerts": [
      "mqtt"
    ],
    "errors": [
      "mqtt"
    ],
    "V": [
      "logger",
      "mqtt"
    ],
    "connectivity": [
      "mqtt"
    ]
  },
  "tls": {
    "server_cert": "/data/certs/server_fullchain.pem",
    "server_key": "/data/certs/server_key.pem"
  },
  "mqtt": {
    "broker": "mosquitto:1883",
    "client_id": "fleet-telemetry",
    "topic_base": "tesla",
    "username": "<redacted>",
    "password": "<redacted>",
    "qos": 1,
    "retained": false,
    "connect_timeout_ms": 30000,
    "publish_timeout_ms": 10000,
    "disconnect_timeout_ms": 2500,
    "connect_retry_interval_ms": 10000,
    "keep_alive_seconds": 30
  }
}
```

Does anything about this jump out as obviously wrong?

agbpatro, 2025-01-16:
Hi @andrew-kennedy 
I suspect something might be wrong in the setup even though other containers are reachable. I have added a healthcheck on server startup which ensures connectivity to MQTT broker is there https://github.com/teslamotors/fleet-telemetry/pull/305. I can't see anything wrong in the config file you pasted.
Also note that the integration test is able to validate the MQTT implementation of the service hence I suspect something is incorrect in your setup itself

andrew-kennedy, 2025-01-17:
What about the mqtt broker config? Anything in particular that needs to be set up a certain way?

agbpatro, 2025-01-17:
If I would have to guess, most likely `"broker": "mosquitto:1883"` is not resolvable. Maybe provide an IP of the MQTT broker instead?

andrew-kennedy, 2025-01-18:
It was actually the way I was running it in a container, I had a typo in the fleet server's container's network name that lead to the broker being inaccessible. The connection attempts I saw in the brokers logs were just coincidentally happening from other containers at the same time.

closed 2025-01-18 by andrew-kennedy