teslamotors/fleet-telemetry #309: Add documentation for `prefer_typed` field

rawmean, 2025-01-16:
Thanks for adding the `minimum_delta` attribute. Since this field requires setting the undocumented `prefer_typed` being set to `true`, can you please document what `prefer_typed` actually does so that we can change our parsing code accordingly?

rawmean, 2025-01-17:
> Instead of returning everything as a string the data is returned typed, so as int ,bool, float, string

I was hoping that was the case, but I'm afraid there is more: 
https://github.com/teslamotors/fleet-telemetry/issues/256

chavezoz, 2025-01-19:
@rawmean , where can I find details on how to set _minimum_delta_?

chavezoz, 2025-01-19:
All good, found it under announcements.

rawmean, 2025-01-19:
> All good, found it under announcements. 

I wish they sent email to registered developers for this sort of things.

patrickdemers6, 2025-01-22:
Added in announcements and the fleet_telemetry_config create endpoint.

closed 2025-01-22 by patrickdemers6