teslamotors/fleet-telemetry #310: MQTT best practice (time stamp / CreatedAt)

Adminius, 2025-01-17:
Hey there!
I'm trying to implent logger for my local telemetry instance with MQTT.
It working fine, but I'm missing timestamps (CreatedAt)

My problem is: my car is asleep since 5h, I'm restarting the logger, it reads al the old data from MQTT and uses "now" as time stamp. -> wrong, because the data is 5h old.

It there any best prcatice for this?

Or maybe we can have possibility to get "original" data json directly in topics?

e.g. telemetry/XP7YGCEK9PB00000/v/ACChargingPower
```
	{
		"ACChargingPower": {
			"stringValue": "6.400000095367432"
		},
		"CreatedAt": "2025-01-16T09:08:39Z",
		"Vin": "XP7YGCEK9PB00000"
	}

```

or may be shorter way:
`{ "stringValue": "6.400000095367432"; "CreatedAt": "2025-01-16T09:08:39Z" }`

this could be optional to keep current implementation compatible.

@erwin314 , @patrickdemers6  what do you think?

agbpatro, 2025-01-17:
I think this is a good idea. Was reading a bit about how to do this via some headers. It seems we use `github.com/eclipse/paho.mqtt.golang` from  [eclipse-paho/paho.mqtt.golang](https://github.com/eclipse-paho/paho.mqtt.golang) . However, if we switch to [eclipse/paho.golang](github.com/eclipse/paho.golang), we can publish UserProperties, which are like [headers](https://stackoverflow.com/a/64908193)

Also, feel free to submit a PR to propagate the timestamp with your proposal. Happy to review them

Adminius, 2025-01-17:
All know about go is that it has two letter "g" and "o" in the name 🤦‍♂️
That's why I'm asking for help 😉

erwin314, 2025-01-19:
> Hey there! I'm trying to implent logger for my local telemetry instance with MQTT. It working fine, but I'm missing timestamps (CreatedAt)
>
> My problem is: my car is asleep since 5h, I'm restarting [...]
...

Hi! It sounds like you're running into one of the key aspects of MQTT: it's simply a protocol for passing messages, not for managing timestamps or ensuring timing integrity of the data. The issue you're facing arises because MQTT does not inherently include time metadata—it's up to the publisher or the system processing the message to handle that.

Since you're working with time-sensitive data, it's important that your logger remains online and actively processes messages in real-time. (So it's your logger thats becomes responsible and MQTT is just passing messages along).

Ideally, the telemetry server should add a timestamp (CreatedAt) as a property (as mentioned by @agbpatro). This way, the logger can use the provided timestamp. Allowing retained MQTT messages to be used as a 'message storage'.

Adminius, 2025-01-19:
Yeah, you are right. I'll try to switch to ZMQ or Kafka instead.

closed 2025-01-19 by Adminius