teslamotors/fleet-telemetry #313: Car stops streaming until restarted

rawmean, 2025-01-18:
Car stops streaming for no apparent reason. 
Waking the car does not fix the problem. 
Sending the configuration to the car does not solve the problem.
Only restarting the car seems to make the car start streaming again.

This, obviously, poses a problem

patrickdemers6, 2025-01-22:
@rawmean can you create an issue in the support portal which includes your VIN and a timestamp?

stx, 2025-01-27:
@patrickdemers6 Just saw a similar issue and snagged the VIN and timestamps. Check ticket SUPPORT-148.

rawmean, 2025-01-29:
I also filed a bug report with requested info: SUPPORT-160

rawmean, 2025-01-29:
More info about this problem:
Sending the config to the car works and response indicates that it has been updated. But when I query the config from the car, it responds with `config: null` (see below):

Response when I send the config:
```vin = REDACTED Telemetry configuration updated successfully. Response data string: {"response":{"updated_vehicles":1}}```
Response when I try to get the config:
```vin = REDACTED Telemetry configuration read successfully. status code: 200 Response data string: {"response":{"synced":true,"config":null <Response [200]>}}```

rawmean, 2025-01-30:
> I see the same behavior with a car that has too many Telemetry integrations. Could that be the case here?

Yes. That's the issue. The current limit according to the documentation is 3. 

https://developer.tesla.com/docs/fleet-api/fleet-telemetry#configuring-a-vehicle

The problem is that the response to the config message is exactly the same as the successful case. Also, even if the user removes one of the apps to go below the limit of 3, the config that you tried to install wouldn't take effect. I guess you have to keep sending the configuration and querying to make sure it has actually taken effect. 

I hope Tesla changes this so that at least the response indicates the error when there are too many apps.