teslamotors/fleet-telemetry #315: 500 on partner account register

jlestel, 2025-01-21:
Hello,

Since yesterday (20/01) around 19:00 UTC, I have 500 errors on partner account creation.

Server error: `POST https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts` resulted in a `500 Internal Server Error` response: {"response":null,"error":"An unexpected error has occurred - a03a67fe99e4ef542f2cd878650253d6","error_description":"","txid" (truncated...)

Maybe there is a limit on the number of applications that can be registered with a root domain (myteslamate.com)? I have dozens of customers who have successfully created them today...

@ThomasAlxDmy I have sent an email to support too (fleetapisupport@tesla.com). Thanks for your feedback as this is a pain and I have many customers waiting for their partner account to be validated.

Geisenpeter93, 2025-01-21:
same issue here. Looking forward to the solution as I need the API for my home automation.

agbpatro, 2025-01-21:
Hi there. We have identified the issue and are looking into it. Fix will be out soon

jlestel, 2025-01-21:
It's working again now.

THANKS!

closed 2025-01-21 by patrickdemers6