teslamotors/fleet-telemetry #317: BatteryHeaterOn always returns invalid

bperlman, 2025-01-21:
It seems that BatteryHeaterOn always reports {"invalid"=>true} in all cases - possible bug?

Bre77, 2025-01-21:
I have also confirmed that `BatteryHeaterOn` is always invalid for my users.

agbpatro, 2025-01-21:
Hi There
We will investigate the issue internally and get back to you.

patrickdemers6, 2025-01-22:
@Bre77 @bperlman I believe this field is only for SX. Creating a ticket to bubble up the equivalent on other vehicles.

bperlman, 2025-01-22:
@patrickdemers6 Thank you, that would be helpful - was operating under the assumption that this telemetry field was equivalent to  `climate_state.battery_heater` from `vehicle_data` which seems to be reporting based on vehicle type equivalent as you mention as far as I can tell.

bperlman, 2025-02-19:
Hey @patrickdemers6 - is there an update/timeline for this to be in the wild? (Just because it's not possible to track this any longer without using vehicle_data until this is change is made) - thanks!

patrickdemers6, 2025-03-03:
This will be resolved in 2025.8.3, fleet streaming client version 1.0.1. Thanks for your patience.

closed 2025-03-03 by patrickdemers6