teslamotors/fleet-telemetry #318: are reliable acks enabled by default?

jbanyer, 2025-01-22:
I'm experiencing lost signals (which I'll raise another ticket for), and am trying the reliable acks as a possible mitigation.

However I seem to get the following log messages regardless of whether `reliable_ack_sources` is set in server_config.json or not.

```
tesla-fleet-telemetry    | {"context":"fleet-telemetry","level":"debug","msg":"message_respond","record_type":"V","response_type":"ack","time":"2025-01-22T00:46:41Z","txid":"78ae54a58800468194dc95-000000397"}
```

Are reliable acks enabled by default?

patrickdemers6, 2025-01-22:
If reliable acks are not enabled for any data providers, the reliable ack is sent immediately.

Unfortunately - and we have a documentation gap here - the vehicle does not change behavior (i.e. resend a message) based on reliable acks. Creating a ticket internally to properly handle acks.

dirkbonhomme, 2025-01-22:
Is there a use case today for enabling reliable ack if it doesn't change the car behavior?

patrickdemers6, 2025-01-22:
No reason to enable, though I'd recommend it as there's no harm. Future proofing your setup for when the vehicle properly handles reliable acks.

patrickdemers6, 2025-03-13:
As of 2025.2.6 / 2024.45.32.20, reliable acks are used for V records when `delivery_policy` is set to `latest`.