teslamotors/fleet-telemetry #319: occasional lost signals

jbanyer, 2025-01-22:
At scale, we are experiencing occasional lost signals.

For example, a vehicle may be unplugged (`DetailedChargeState = Disconnected`), and then we start receiving signals showing that the vehicle is clearly charging, eg `ACChargingPower` goes to non-zero, `TimeToFullCharge` gets a value etc, but there is no update to `DetailedChargeState` until much later eg when charging is complete.

We've also seen the reverse, where `DetailedChargeState` changes to `Charging` but there is no update of `ACChargingPower` until much later.

Estimated occurrence rate is around 1 in 200 vehicles in a day, although it's hard to detect all situations.

Our architecture is:

vehicle -> AWS NLB -> fleet-telemetry -> Kafka (AWS MSK) -> application

- There is only a single instance of each component in the architecture, although Kafka has 2 brokers.
- MSK stats report zero network drops and resource utilisation is very low (CPU and disk).
- Reliable acks are not explicitly configured for fleet-telemetry, although as per #318 the logs suggest they may be enabled?
- fleet-telemetry prometheus metrics are being gathered:
  - kafka queue size is between 1 and 5
  - kafka acks equal kafka produce
  - kafka produce is always slightly higher than `dispatch`, with one noticeable spike, see graph. That seems the wrong way around if there was messages being dropped?

Any advice would be greatly appreciated, as this is a very nasty issue to deal with and causes major malfunctions in any automation app. @patrickdemers6 

I'm aware that the `resend_interval_seconds` parameter would help alleviate the impact of this issue (without actually solving it), but the 
majority of our vehicles do not yet support it (they are mostly in Australia). We'll use it when we can.

[image]
This graph shows the constant difference between `dispatch` and `kafka produce` metrics from fleet-telemetry. Why is `kafka produce always slightly higher than `dispatch`?

patrickdemers6, 2025-01-22:
This is a known, infrequent issue. No luck finding the root cause yet, but this is a high priority.

marklindsey, 2025-01-22:
I have encountered this, too, though not enough to detail potential root cause or reproduction steps (2024 Model 3 LR RWD)

patrickdemers6, 2025-01-22:
I'm investigating this as if it's a vehicle-side issue. @jbanyer have you noticed if any subset of vehicles is disproportionately impacted? Vehicle model, year, uses wifi, etc

bperlman, 2025-01-22:
@patrickdemers6 - can confirm that we're seeing this happening on a pretty widespread basis on critical fields that are necessary to monitor for state change accuracy and to compute drive/charge characteristics such as `Gear`, `DetailedChargeState`, and `RatedRange` it's a little bit tricky to detect at scale until we see the result of it, but it happens more often than not on one of team members' 2021 Model 3, version 2024.45.32.1 ee93da10a358, switches between wifi/cellular when arriving/departing home (not sure if any of that helps - can send a VIN privately if helpful).

marklindsey, 2025-01-22:
That's very similar to my situation @bperlman. Perhaps the car drops its cellular signal and sends messages before wifi connects leaving those in the abyss.

bperlman, 2025-01-22:
Just to be clear, we are seeing this happen regardless of if a wifi/cellular switch is taking place at the time, but, flagging that this particular vehicle does regularly utilize both connection types just in case means anything at the vehicle level (i.e. compared to a vehicle that rarely connects to wifi).

marklindsey, 2025-01-22:
Agreed that this is intermittent. In this same use case where my car is temporarily offline, I have also had the telemetry records queued and then sent en masse once the car reconnected.

jbanyer, 2025-01-22:
> I'm investigating this as if it's a vehicle-side issue. [@jbanyer](https://github.com/jbanyer) have you noticed if any subset of vehicles is disproportionately impacted? Vehicle model, year, uses wifi, etc

@patrickdemers6 at this stage I have not been able to look for any correlations with firmware version, vehicle type etc.

It's a difficult problem to analyse because it's very difficult to detect. Fleet telemetry often results in temporary inconsistencies between fields values because they are sent as separate signals, eg ACChargingPower may increase but ChargeAmps is still zero for a short time. I have tried to report on this problem by looking for sustained periods where the state is inconsistent. I'm sure it must be happening more than I've found with this method.

patrickdemers6, 2025-01-23:
@jbanyer there is a sequence number! If you inspect the txid of the message, it is formatted as `{uuid}-{counter}`.

stx, 2025-01-23:
I've already been working with @patrickdemers6 and some others at Tesla for awhile on this, but wanted to publicly +1 it and share what we're seeing so we can get to the bottom of it. This is a mission critical issue that causes major problems with tracking, software usability and vehicle controls.

* Have witnessed this behavior on 2021-2024 Model 3/Y.
* Have **not** witnessed this behavior on any Model S/X, Cybertruck or Semi. Unclear whether this is due to platform differences, or just that these vehicles aren't as common and so it hasn't been detected yet.
* It's unclear which fields are affected. Some fields _seem_ to be more affected than others, but I'm guessing that's just because those fields are more critical and thus easier to detect issues with. Fields that we've verified to be impacted: 
  * DetailedChargeState
  * DoorState
  * Locked
* _Seems_ to be worse when connectivity is poor or changing. We repeatedly moved a test vehicle closer to and further from a WiFi router, and the issue _seemed_ to get worse when it was further away.

Would be good if someone can verify if they've seen this problem on Model S/X, Cybertruck or Semi.

stx, 2025-01-24:
Just had a car drop a bunch of different fields at the same time. So it's not just a random signal issue - something is causing all data to get dropped intermittently.

marklindsey, 2025-01-24:
> [@jbanyer](https://github.com/jbanyer) there is a sequence number! If you inspect the txid of the message, it is formatted as `{uuid}-{counter}`.

It's pretty easy to identify which record.Txid is being dropped now. Is there any way to request these specific records from the vehicle? Are they saved temporarily?

Here's another example of data being dropped
`4:38:38 am | V | 4210 | {"EstimatedHoursToChargeTermination":"<invalid>"}`
`4:37:28 am | V | 4208 | {"DetailedChargeState":"Complete","Soc":79.97}`

bperlman, 2025-01-24:
Similar experience. Upon looking at this much more closely dropped signals are more the norm than the exception (in my experience anyway with our test vehicles). I'll echo @stx that this is a mission-critical issue as it makes it all but impossible to reliably track activity such as driving/charging using fleet telemetry at this time. 

From what I can tell, it happens most often around vehicle sleep/wake changes and connectivity changes. During an example this morning, Fleet API's `vehicles` endpoint continued to show this vehicle as `offline`, however it started to stream telemetry. From the sequence numbers, we can see that several values were dropped. In this particular case, it sent a `DetailedChargeState` change to `DetailedChargeStateCharging` but it appears that the corresponding end of the charge was dropped.

Here's an example from today that may help. This is a 2022 Model Y, `2024.45.25.15 2409d648f093` on WiFi.

- Vehicle was reporting offline via Fleet API `vehicles` endpoint (Model Y sleeping but on WiFi so often shows as offline) as of `2025-01-24 10:42:02.114`. Confirmed every minute between then until `2025-01-24 13:22:36.890 UTC`, vehicle continued to report state as `offline` for that entire duration.
- It was set to auto pre-condition at `2025-01-24 12:00 UTC`
- Vehicle started reporting telemetry at `2025-01-24 11:44:16.641 UTC`, starting with `location` (position has not changed in several days)
- Vehicle reported telemetry value `2025-01-24 11:56:54.644 UTC` `DetailedChargeState` `{"detailedChargeStateValue": "DetailedChargeStateCharging"}` 
- Inspecting the `txid`s, can see that it skipped 7 signals between `2025-01-24 12:02:30.644 UTC` and `2025-01-24 12:05:28.645',  then again skipped 8 signals between `2025-01-24 12:09:24.644 UTC` and `2025-01-24 12:20:13.645`, then skipped 3 signals between `2025-01-24 12:38:43.645 UTC` and `2025-01-24 12:40:00.645`

Fleet API reported the vehicle as offline this entire time - so not sure if that gives any clues. 

The effect of this is there's a signal showing a charge starting, but, there is no signal showing an end (and still isn't as of the time that I'm writing this). Dropping a less-critical signal that doesn't signify a state change isn't a huge deal as it's fairly likely to change again shortly, but since something like `DetailedChargeState` may not change for several hours after, it goes without saying how problematic that can be.  FWIW, solving the root cause aside, it seems like having acks be functional would help with this issue in the future.

marklindsey, 2025-01-24:
Is additional ack functionality necessary? From my understanding of web sockets, each message should already have a built in reply acknowledgement from the server implying the websocket connection is still present. If the server didn't respond successfully, it may be worth resending the message.

bperlman, 2025-01-24:
@marklindsey - lack of clarity on my part, I was referring to how they aren't actually actioned by the vehicle yet: https://github.com/teslamotors/fleet-telemetry/issues/318

bperlman, 2025-01-24:
> Would be good if someone can verify if they've seen this problem on Model S/X, Cybertruck or Semi.

@stx -  based on some tests I’ve just done - issue is apparent and widespread on S3XY/Cybertruck with pretty much uniform frequency (distribution of issues matches distribution of vehicle types). Have shared some numbers directly with Tesla for now.

Bre77, 2025-01-25:
I started collecting and analysing the txid after I saw this issue and this is what I can tell so far from my customers:

On a per-vehicle basis over 24 hours:
Average loss = 4.5 messages lost
95% percentile = 7 messages lost
99% percentile = 25 messages lost 
Max = 706 messages lost

marklindsey, 2025-01-25:
Does it make sense to allow third-party apps to request missing txids in the interim as a workaround until this is solved?

marklindsey, 2025-01-30:
Has anyone noticed any relation with fields being dropped and having a lower value for interval_seconds configuration?

EmilOlovsson, 2025-02-01:
I'm also losing signals randomly, is there any way to get reliable acks?

jbanyer, 2025-02-07:
@patrickdemers6 I found something that might be relevant.

My system now looks for inconsistent telemetry fields, in particular when DetailedChargeState != Charging but ACChargingPower is non-zero (or amps etc).

When the fields are inconsistent for 5 minutes, it performs a vehicle_data poll.

I have found that many of these polls also return the same inconsistent data.

The logs below show polled data converted into telemetry fields (I don't have the raw polled data unfortunately, but I'm confident it agrees)

```
2025-02-06 00:00:22.684 DEBUG correctInconsistentFields corrected fields: {"ChargeAmps":10,"Version":"2024.44.25.4 ad52e3119f65","ACChargingEnergyIn":4.7,"DCChargingPower":0,"ACChargingPower":2.23,"Soc":100,"ChargeCurrentRequestMax":10,"ServiceMode":false,"ScheduledChargingStartTime":"Thu Feb 6 00:15:00 2025","ChargeCurrentRequest":10,"ChargeLimitSoc":100,"VehicleName":"White Angel","ChargePortLatch":"Engaged","ChargerPhases":1,"DetailedChargeState":"DetailedChargeStateComplete","TimeToFullCharge":0.5,"IdealBatteryRange":256.63,"ScheduledChargingMode":"StartAt","lastSeenAtMs":1738760420474}
2025-02-06 00:03:37.947 DEBUG correctInconsistentFields corrected fields: {"IdealBatteryRange":233.52,"Version":"2024.45.25.5 bb0c321a99db","ScheduledChargingMode":"Off","ChargeLimitSoc":85,"ACChargingEnergyIn":6.26,"DCChargingPower":0,"VehicleName":"Carlo","ChargeCurrentRequest":48,"TimeToFullCharge":0.08333333333333333,"Soc":85,"ServiceMode":false,"ChargeCurrentRequestMax":48,"ChargePortLatch":"Engaged","DetailedChargeState":"DetailedChargeStateStopped","ScheduledChargingStartTime":null,"ChargerPhases":1,"ChargeAmps":14,"ACChargingPower":3.22,"lastSeenAtMs":1738760617206}
2025-02-06 00:04:20.894 DEBUG correctInconsistentFields corrected fields: {"ScheduledChargingStartTime":"Wed Feb 5 07:00:00 2025","DCChargingPower":0,"VehicleName":"El Diablito","ScheduledChargingMode":"StartAt","TimeToFullCharge":1,"ChargeAmps":4,"ChargeCurrentRequest":23,"ChargeLimitSoc":80,"ACChargingEnergyIn":0,"Version":"2024.45.25.5","ACChargingPower":0.98,"IdealBatteryRange":252.92,"Soc":75,"ServiceMode":false,"ChargeCurrentRequestMax":40,"ChargePortLatch":"Engaged","ChargerPhases":1,"lastSeenAtMs":1738760660060,"DetailedChargeState":"DetailedChargeStateStopped"}
2025-02-06 00:04:35.381 DEBUG correctInconsistentFields corrected fields: {"ScheduledChargingMode":"Off","ServiceMode":false,"ChargeCurrentRequest":12,"IdealBatteryRange":216.61,"ScheduledChargingStartTime":null,"DCChargingPower":0,"TimeToFullCharge":7.75,"Soc":83,"ChargeCurrentRequestMax":12,"VehicleName":"Baymax","ChargePortLatch":"Engaged","ChargerPhases":1,"ACChargingPower":0.48,"ChargeLimitSoc":100,"ACChargingEnergyIn":0,"ChargeAmps":4,"Version":"2025.2 cc1c1ea25d1f","lastSeenAtMs":1738760674912,"DetailedChargeState":"DetailedChargeStateStopped"}
```

stx, 2025-02-07:
It's normal for a Tesla to pull charger power while not charging. That's how preconditioning works.

jbanyer, 2025-02-07:
> It's normal for a Tesla to pull charger power while not charging. That's how preconditioning works.

Ah yes, sorry, I keep forgetting about that. I'll delete my comment above.

The logic we use to detect inconsistent telemetry is actually: `DetailedChargeState != Charging && TimeToFullCharge > 0`. It's based on observations that show `TimeToFullCharge` is zero or unset when not charging, including when pulling power for preconditioning.

But this condition is triggering a lot, so I guess it's actually an expected condition?

stx, 2025-02-07:
There's a race condition problem with fleet telemetry data unrelated to this issue which can cause issues like this. Assume interval @ 60s:

00:00 - Car starts charging, sends DetailedChargeState == charging
00:03 - Car calculates charge time, sends TimeToFullCharge == 1 hour
00:50 - Car finishes charging
01:00 - 60s passes, car sends DetailedChargeState == complete
... Now the car is not charging and TimeToFullCharge > 0 ...
01:03 - 60s passes, car sends TimeToFullCharge == 0/

jbanyer, 2025-02-07:
> There's a race condition problem with fleet telemetry data unrelated to this issue which can cause issues like this. 

Yes sorry, that's why we only detect inconsistent data if it's been inconsistent for over 5 minutes. I'll update my comment.

marklindsey, 2025-03-06:
For the most part, this seems to have subsided with the delivery_policy "latest", however, I have still encountered lost signals post configuration during a software update. Not sure whether or not this is intended as I could imagine during a software update the buffer might be emptied and the vehicle may fluctuate between offline and online.

patrickdemers6, 2025-03-06:
Great to hear. During a software update, it's completely expected to lose signals. Parts of the vehicle will be periodically offline and may experience frequent reboots. Unfortunately there is no easy solution for this case.

andrewdevelopz, 2025-03-06:
Hey @marklindsey can you please clarify what you mean by when setting `delivery_policy "latest"`? 

Are you referring to a setting within fleet-telemetry or maybe a setting within one of the supported [dispatchers](https://github.com/teslamotors/fleet-telemetry?tab=readme-ov-file#personalized-backendsdispatchers)?

jbanyer, 2025-03-07:
@patrickdemers6 after moving our users' vehicles to a new telemetry configuration, preliminary data suggests that the lost signal issue has significantly improved.

[image]

The new configuration included the following changes:
- delivery_policy = latest
- resend_interval_seconds=300 for DetailedChargeState

I haven't looked too deeply at the logs to try to determine which of these changes has made the most difference.

Can you explain what the `is_resend` flag indicates on `tesla_V` records? Most the records with `is_resend` set appear to have most or all fields set.

patrickdemers6, 2025-03-13:
`is_resend` indicates the message is a resend. All fields that have had changes that haven't been acked are included.

Glad the changes seem to have drastically improved the issue. Going to go ahead and close this.

closed 2025-03-13 by patrickdemers6