teslamotors/fleet-telemetry #320: Feature Request: Signal dependencies

chavezoz, 2025-01-22:
Hi Tesla Team,

Is there anything on your roadmap in regards to dependencies between signals? I.e. send additional data in the same payload when a parent signal event evaluates as true.

As an example, let's say I'm only interested in tracking the vehicles current speed, speed limit and coordinates when an Emergency Lane Departure Avoidance event is recorded.

 "fields": {
			"EmergencyLaneDepartureAvoidance": {
				"interval_seconds": 20,
				"dependancies":[
					"Location",
					"VehicleSpeed",
					"Location",
					"CurrentLimitMph"	
				]
			}
    }

In certain use cases, this would significantly reduce traffic from the vehicle to backend servers instead of purely relying on the "interval seconds" model.

patrickdemers6, 2025-01-22:
Similar functionality has been proposed in the past but wasn't a priority at the time. I'm not opposed to supporting this concept though!

marklindsey, 2025-01-22:
+1. On a semi related topic, I get several redundant responses for fields like Location when it's parked or charging. The latitude or longitude generally differs by 0.000001 or 0.000002.

corsair, 2025-05-05:
Has there been any updates or consideration for this, @patrickdemers6? We run into common situations where a charging session will begin and we'll get the values for ChargeVoltage, ChargeRateMilePerHour, etc. before we receive the value of ChargeState or DetailedChargeState, even when all fields are set to the same interval.

Being able to have the vehicle always send the charge states alongside relevant charging values would be incredibly useful to ensure all received (and stored) values are consistent with the vehicle's live state.