teslamotors/fleet-telemetry #327: Incorrect door state

period, 2025-01-25:
Haven't had the opportunity to test with the remaining doors, but I noticed that the incorrect door was showing as open. Front passenger door opened, but DriverRear was showing as true on the DoorState signal.

Based in the UK - so RHD, car is running 2024.45.42.2. The app shows the doors correctly.

stx, 2025-01-25:
We've seen the same behavior which is due to https://github.com/teslamotors/fleet-telemetry/issues/319.

period, 2025-01-25:
> We've seen the same behavior which is due to [#319](https://github.com/teslamotors/fleet-telemetry/issues/319).

Admittedly I've only skim read the above issue as out and about, but I'm not sure if this is related to a signal being lost - I received the signal for the door being opened, then the signal for the door being closed when it was subsequently closed a few seconds later - so the amount of signals I was expecting to receive was correct, but the data contained in both of them was for the wrong door.

Reviewing all the txids I received, they seem in sequence with none missing.

Forgot to mention but car is a 2020 M3

sjhill01, 2025-01-25:
Came to look for known issues with DoorState as I'm just getting a null string when a door is opened?

patrickdemers6, 2025-01-28:
Thanks for the report, surprised this hasn't been caught earlier! Looks like the proto fields were swapped.

This is different from #319 as all signals are being delivered.

closed 2025-01-28 by patrickdemers6

period, 2025-01-28:
Thank you @patrickdemers6 !

Bre77, 2025-01-28:
Fixing this has broken it on my vehicle. Clearly this issue is not widespread and is specific to certain firmware or vehicle types.

https://github.com/teslamotors/fleet-telemetry/issues/336

PR #333 (2025-01-28): fix proto mixing up passenger front and driver rear
# Description

These two doors were accidentally swapped. They will remain messed up for `prefer_typed:false`.

Fixes #327 

## Type of change

Please select all options that apply to this change:

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] Documentation update

# Checklist:

Confirm you have completed the following steps:

- [ ] My code follows the style of [... cut]