teslamotors/fleet-telemetry #332: Issue with minimum_delta for Location

EmilOlovsson, 2025-01-28:
**How do I implement minimum_delta correctly for "Location"?**

The vehicle is a Model 3 running software V12 (2024.45.32.2)

I've set minimum_delta set to 0.0001 for Location but still receive data that has very small changes:

{"data":[{"key":"Location","value":{"locationValue":{"latitude":YY.609382,"longitude":XX.099705}}}],"createdAt":"2025-01-28T17:04:28.921265359Z","vin":"XXX"}

{"data":[{"key":"Location","value":{"locationValue":{"latitude":YY.609382,"longitude":XX.099704}}}],"createdAt":"2025-01-28T17:06:29.234057117Z","vin":"XXX"}

patrickdemers6, 2025-01-28:
Location doesn't support minimum_delta yet, we will be shipping that in a new release soon.

patrickdemers6, 2025-01-28:
It will be in meters, so if you pass min delta of 10, it will send when location changes by 10+ meters.

rawmean, 2025-01-29:
can you please consider stop charging for API before a new firmware that supports minimum_delta is pushed to majority of cars? Otherwise, we are charged for data that we have not requested?

marklindsey, 2025-01-30:
Very happy to hear this @patrickdemers6! Will this require a telemetry server code change also or simply a vehicle telemetry config change?

fredli74, 2025-02-01:
> It will be in meters, so if you pass min delta of 10, it will send when location changes by 10+ meters.

Odd, I expected it to be in decimal degrees (that's how I implemented it). I get it, "location" is a compound of latitude and longitude. At least I'm happy you went with meters as opposed to the "odometer" values. 😉

EmilOlovsson, 2025-05-24:
@patrickdemers6 Is this fixed?

According to the fleet telemetry changelog this should now work. https://developer.tesla.com/docs/fleet-api/fleet-telemetry#changelog

- Location fields now support minimum_delta. Changes in distance are measured in meters.