teslamotors/fleet-telemetry #335: Also support metric besides imperial

netdata-be, 2025-01-28:
Most of the fields regarding distances are in miles, can you also support metric.

Adminius, 2025-01-31:
Uhm, odometer is also in miles...

thefireblade, 2025-02-14:
We're not planning on changing this behavior, this can be computed anyways

closed 2025-02-14 by thefireblade

netdata-be, 2025-02-17:
@thefireblade I can understand, but it's confusing at least that the `SettingDistanceUnit` is reporting `DistanceUnitKilometers` while in fact it are miles.

So if the idea is not to change the units, then please fix the `SettingDistanceUnit` so that we can use this field to convert the values.

Currently it is a mix between imperial and metric...

Bre77, 2025-02-17:
> @thefireblade I can understand, but it's confusing at least that the `SettingDistanceUnit` is reporting `DistanceUnitKilometers` while in fact it are miles.

That signal tells you what unit of measurement the vehicle is configured to display, it's behaving correctly, and doesn't change that Fleet Telemetry continues to report the vehicle's native imperial units of measure.