teslamotors/fleet-telemetry #336: Door positions were broken

Bre77, 2025-01-28:
PR #333 has broken the previously working door positions on my  China built 2022 RHD Model 3 (LRW3F7EK4NCxxxxxx) running 2024.44.25

`prefer_typed: true`

```json
{
  "publish_time": 1738103424.99,
  "data": {
    "data": [
      {
        "key": "DoorState",
        "value": {
          "doorValue": {
            "DriverFront": true,
            "DriverRear": false,
            "PassengerFront": true,
            "PassengerRear": false,
            "TrunkFront": false,
            "TrunkRear": false
          }
        }
      }
    ],
    "createdAt": "2025-01-28T22:30:24.875285416Z",
    "vin": "LRW3F7EK4NC716336"
  },
  "attributes": {
    "receivedat": "1738103424000",
    "txtype": "V",
    "timestamp": "0",
    "txid": "8f321eff87e345f5815a5b-000000011",
    "version": "0",
    "vin": "LRW3F7EK4NCxxxxxx"
  }
}
```

[image]

period, 2025-01-29:
My VIN (which appears to have started this) for reference if it helps: 5YJ3F7EB8LFxxxxxx , 2020 RHD M3LR dual motor running 2024.45.32.2 7cc287a111a2

patrickdemers6, 2025-01-29:
A firmware regression which swapped front passenger and driver rear doors was included in 2024.44.32. The fix from this morning (#333) swapped the subset of vehicles with door positions messed up.

Vehicles running firmware versions prior to 2024.44.32 are now swapped, but going forward this is stabilized.

I'll leave this open as a PSA to those noticing this unexpected breaking change.

Bre77, 2025-01-29:
So to be clear @patrickdemers6 there was a firmware regression, but that regression is being left in and the way the proto is now is how it will stay moving forward?

patrickdemers6, 2025-02-12:
Yes, will keep it as is moving forward.

closed 2025-02-12 by patrickdemers6