teslamotors/fleet-telemetry #337: minimum_delta option not working as expected

andrewdevelopz, 2025-01-29:
vehicle firmware version: 2024.45.25
fleet-telemetry version: v0.6.0

I have a fleet-telemetry config created as below

```
{
    "response": {
        "synced": true,
        "config": {
            "prefer_typed": true,
            "fields": {
                "PackCurrent": {
                    "interval_seconds": 60,
                    "minimum_delta": 0.2
                },
                "PackVoltage": {
                    "interval_seconds": 60,
                    "minimum_delta": 1
                }
            }
        }
    }
}
```

I am expecting to receive these fields only if the value changes based on the minimum_delta. However I am receving the fields every 60 seconds, even if the value did not change at all.

Example log:
```
{
    "data": {
        "CreatedAt": "2025-01-28T06:19:07Z",
        "PackCurrent": -0.7000000104308128,
        "PackVoltage": 382.22999145649374,
    },
    "metadata": {
        "txid": "de9a74cb3a9245f4bf5502-000000001",
    },
}
{
    "data": {
        "CreatedAt": "2025-01-28T06:20:07Z",
        "PackCurrent": -0.7000000104308128,
        "PackVoltage": 382.1799914576113,
    },
    "metadata": {
        "txid": "de9a74cb3a9245f4bf5502-000000002",
    },
}
{
    "data": {
        "CreatedAt": "2025-01-28T06:21:07Z",
        "PackCurrent": -0.7000000104308128,
        "PackVoltage": 382.20999145694077,
    },
    "metadata": {
        "txid": "de9a74cb3a9245f4bf5502-000000003",
    },
}
```

Is this a bug or am I doing something wrong on my side?

Adminius, 2025-01-31:
same here, FW 2024.45.32.2
PackVoltage seems to work, PackCurrent not

[image]

[image]

patrickdemers6, 2025-02-12:
@andrewdevelopz this is expected for 2024.45.25 as it doesn't have the required firmware version of 2024.44.32 / 2024.45.32.

@Adminius I suspect the sensor flaps the value it reports and your minimum delta is smaller than the flap. Can you try setting PackCurrent's minimum delta to a slightly larger value?

andrewdevelopz, 2025-02-12:
@patrickdemers6 isn't my vehicle version `2024.45.25` later than version `2024.44.32`?

closed 2025-02-14 by thefireblade