teslamotors/fleet-telemetry #341: Charging hardware data getting invalidated when vehicle goes to sleep

stx, 2025-01-31:
Right before vehicles go to sleep, they’re passing “invalid” for several hardware-related fields, including:

* FastChargerPresent
* ChargingCableType
* ChargePortDoorOpen
* ChargePortLatch

This is a problem on FastChargerPresent and ChargingCableType because “invalid” also means the charge cable has been disconnected.

As a result, the car is telling us the cable is disconnected when it’s not. We would expect FastChargerPresent and ChargingCableType to only go "invalid" when physically disconnected.

Also tracking this on support ticket SUPPORT-140.

patrickdemers6, 2025-02-11:
Invalid shouldn't be interpreted as a change to the base/regular state. It means the process on the vehicle that populates its value is no longer reporting anything. This is most likely to happen when the vehicle is going to sleep.

closed 2025-02-11 by patrickdemers6

Bre77, 2025-02-12:
For other sensors it *does* reflect a change to the base/regular state, such as `DestinationName`, `MilesToArrival`, `TimeToFullCharge` etc.

I assume we need to handle this at an application level, such as ignoring invalid on the signals @stx identified, but not on others?

stx, 2025-02-12:
It also reflects a change in the fields I listed. For example, if you unplug the car, ChargingCableType returns invalid.