teslamotors/fleet-telemetry #342: Grafana Datasource

krishR007, 2025-02-03:
Does the Fleet Telemetry store log any dependencies through like Loki , Mongo etc? Can I check logs on a Grafana dashboard using Kubernetes?

Bre77, 2025-02-03:
It has a Prometheus metric endpoint you can enable, and a statsd sender. They repo even has an example Grafana dashboard, and I was at one point monitoring my Fleet Telemetry servers with Grafana Cloud (although that needed a reverse proxy for auth).

krishR007, 2025-02-04:
@patrickdemers6 Can you explain please ?

patrickdemers6, 2025-02-04:
It is possible to get vehicle data showing up in Grafana. I've done it before using the logger datastore and Loki. Then, you have to create a dashboard in Grafana based on Loki queries.

closed 2025-02-04 by patrickdemers6