teslamotors/fleet-telemetry #352: Feature request: profile name

dwery, 2025-02-06:
I'd love to have a telemetry field with the name or id of the key that's currently "driving" the car, it would be useful for tracking/accounting.

Bre77, 2025-02-06:
A slight twist one this, the vehicle data used to provide us the Tesla profile that was driving the car, but I think it was removed and definitely not in Fleet Telemetry. I certainly would get value being able to differenciate which profile is active.

rileymd88, 2025-02-06:
+1 This would be very useful!

corsair, 2025-02-08:
+1, users have requested this for years, and it would help with filtering through driving sessions.

patrickdemers6, 2025-02-14:
Added to an internal ticket tracking field requests.

closed 2025-02-14 by patrickdemers6

rileymd88, 2026-03-15:
Any update on this? I see the info is now available in the app

[image]