teslamotors/fleet-telemetry #354: Feature Request: Configurable Partition Strategy

efriesner, 2025-02-06:
I'd like to be able to guarantee order per vehicle by setting the partitioning strategy to hash instead of the default strategy (least bytes), which would guarantee that all messages for the same vehicle go to the same partition. This seems relatively easy to do as Kafka has a built in balancer option that hashes by the message key, which appears to be already set to the vehicle vin.

patrickdemers6, 2025-02-14:
Reading the code, I'm surprised this isn't happening already. Happy to accept a PR.

elffjs, 2025-02-19:
Maybe I'm misunderstanding you, but: can you do this already? You can set librdkafka settings in the `kafka` section of the config, and we've been setting `"partitioner": "murmur2"`.

efriesner, 2025-02-25:
You are right, will try this out. Thanks!

agbpatro, 2025-03-06:
> [@patrickdemers6](https://github.com/patrickdemers6) it does look like partitioning by VIN is already happening without needing to change the kafka settings. Where is that getting set in the existing implementation?

see all supported kafka settings from https://github.com/teslamotors/fleet-telemetry/blob/main/config/config.go#L61
the default `partitioner` is `consistent_random` which partitions based on key. Key here is `VIN`