teslamotors/fleet-telemetry #355: Public key download failed - error: too large"

chavezoz, 2025-02-08:
Hi all,

Has anyone run into the following error before? 

I'm setting up a new domain for my fleet-telemetry server (initial domain worked fine but need to change it) and have published the public key as  https://mydomain.com/.well-known/appspecific/com.tesla.3p.public-key.pem. When I then call https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts to register the account, I get the following error:

 
`{
    "response": null,
    "error": "Public key download failed for https://mydomain.com/.well-known/appspecific/com.tesla.3p.public-key.pem, error: too large",
    "error_description": "",
    "txid": "92e8f39a5fc4c89c2794f3aa8fc42152"
}`

I'm not sure what the issue is. I didn't run into this issue when setting up the original domain.

Any ideas?

Bre77, 2025-02-08:
Well is your public key like 4 lines long? The error says its too large, possibly you have uploaded the wrong thing. If you shared your domain I could actually check your public key.

chavezoz, 2025-02-08:
All sorted, recreated the public key and now working. Not sure what caused the initial one to not work.

closed 2025-02-08 by chavezoz