teslamotors/fleet-telemetry #375: Determine if FSD / autopilot was enabled during the trip

namelesstwice, 2025-03-26:
Hi all,

I am trying to build some automations with Fleet Telemetry API, specifically I want to understand if autopilot / FSD was used on a specific trip. I can see some fields to indicate lane / speed assistance level in this file:

https://github.com/teslamotors/fleet-telemetry/blob/main/protos/vehicle_data.proto

But I'm not sure if these fields allow to actually determine this. So, I have 2 questions:

1. Is there currently a way to determine if FSD / autopilot was enabled during the trip?
2. If not, are there any plans to implement this functionality?

corsair, 2025-04-23:
CruiseState (which provided access to Full Self-Driving / Autopilot utilization) was removed last year based on the following PR: #236 

I would like to see this decision reversed for the following reasons:

- Full Self-Driving and Autopilot utilization is already presented and made available to vehicle owners enrolled in Tesla Insurance via the Safety Score tab.
- Tesla vehicle owners have requested global access to Safety Score for years, and Elon has previously stated that this would be considered over two years ago.
- The lack of access to this data **encourages the utilization of third-party physical modifications to vehicles**, which, at best, can result in battery drain, and at worst, does not guarantee that a vehicle owner's privacy is maintained.

This feels like a win for privacy, safety, and protecting the vehicle owner's data, which should always be the top priority. The inaccessibility to this field and data risks continued harm and presents a risk for the global fleet.

gbhall, 2025-09-30:
I too would like to see this revisited. 

TezLab claim they can determine FSD engaged status:

https://x.com/tezlabapp/status/1942600505735213508?s=46

If this was made available to one developer it should be opened up to all.

closed 2025-09-30 by aaronpkahn

corsair, 2025-11-25:
Hi @aaronpkahn,

With the latest software update (2025.38.9.5), Tesla is now including statistics on Full Self-Driving utilization. This currently only provides total utilization and is accessible only in-vehicle. Sharing these metrics has required customers to take crude photos of their vehicle screens, which is an inefficient and ineffective way to transfer information. For customers eager to share these metrics, this increases the risk of privacy compromises due to EXIF metadata in most smartphone photos.

The inclusion of these metrics indicates that there is some consideration for providing context of usage to Tesla customers. As such, could there be any reconsideration of reintroducing CruiseState in the near future? Tesla also recently released the [Vehicle Safety Report](https://www.tesla.com/fsd/safety) minisite, which highlights total miles driven (including city miles).

Determining Full Self-Driving utilization is still possible by observing various telemetry fields. However, it is not perfect, expensive in terms of signal throughput, and can lead to misconceptions due to erroneous recordings.

---

Full Self-Driving continues to evolve, and with its eventual expansion into the Tesla Semi, preventing access to this data field is detrimental to the software's growth and to developers' ability to build the best possible experiences.

Thank you.