teslamotors/fleet-telemetry #386: createdAt timestamp is incorrect in initial record after vehicle wake

jbanyer, 2025-05-13:
We've observed that the `createdAt` field is, on occasion, incorrect in the initial record sent by vehicles after they wake up.

In particular, the `createdAt` field contains a value that is in the future, sometimes by as much as 5 minutes, but usually by around 30 seconds or so.

Subsequent messages from that vehicle contain the correct `createdAt`, even if they are sent a split second after the initial message.

Perhaps the clock used to set `createdAt` is incorrect for a brief period after the vehicle wakes up?

This issue is causing nasty problems in our system as we rely on `createdAt` to maintain ordering of updates, and these initial incorrect values cause us to drop subsequent updates. In particular it results in an incorrect `DetailedChargeState` if the vehicle wakes up and immediately starts charging.

(I realise this is probably not an issue with the code in this repo, but figured it's a good one for public visibility and discussion)

dirkbonhomme, 2025-05-14:
Interesting, I believe we are also seeing the side effects of this. Will try to log something as well..

aaronpkahn, 2025-05-16:
We've had a few reports of this and are investigating.  We'll keep you posted on a fix/mitigation.