teslamotors/fleet-telemetry #412: Values chaning back and forth

Urkman, 2025-09-29:
I'm still having a problem with values changing back and forth.
I though this was fix, isn't it?

Log:

> 2025-09-29 06:53:12	energyRemaining:51.23999885469675
2025-09-29 06:52:32	energyRemaining:51.259998854249716
2025-09-29 06:52:02	energyRemaining:51.23999885469675
2025-09-29 06:51:42	energyRemaining:51.259998854249716
2025-09-29 06:51:12	energyRemaining:51.23999885469675
2025-09-29 06:50:52	energyRemaining:51.259998854249716
2025-09-29 06:50:32	energyRemaining:51.23999885469675
2025-09-29 06:49:32	energyRemaining:51.259998854249716
2025-09-29 06:49:12	energyRemaining:51.23999885469675
2025-09-29 06:49:02	energyRemaining:51.259998854249716
2025-09-29 06:48:32	energyRemaining:51.23999885469675
2025-09-29 06:48:22	energyRemaining:51.259998854249716
2025-09-29 06:48:12	energyRemaining:51.23999885469675
2025-09-29 06:47:52	energyRemaining:51.259998854249716
2025-09-29 06:47:32	energyRemaining:51.23999885469675
2025-09-29 06:47:12	energyRemaining:51.259998854249716
2025-09-29 06:47:02	energyRemaining:51.23999885469675
2025-09-29 06:46:52	energyRemaining:51.259998854249716
2025-09-29 06:46:42	energyRemaining:51.23999885469675
2025-09-29 06:46:22	energyRemaining:51.259998854249716
2025-09-29 06:46:02	energyRemaining:51.23999885469675
2025-09-29 06:45:52	energyRemaining:51.259998854249716
2025-09-29 06:45:42	energyRemaining:51.23999885469675
2025-09-29 06:45:12	energyRemaining:51.259998854249716
2025-09-29 06:44:52	energyRemaining:51.23999885469675
2025-09-29 06:44:32	energyRemaining:51.259998854249716
2025-09-29 06:44:22	energyRemaining:51.23999885469675
2025-09-29 06:44:02	energyRemaining:51.259998854249716
2025-09-29 06:43:12	energyRemaining:51.23999885469675
2025-09-29 06:43:02	energyRemaining:51.259998854249716
2025-09-29 06:42:52	energyRemaining:51.23999885469675
2025-09-29 06:42:12	energyRemaining:51.259998854249716
2025-09-29 06:42:02	energyRemaining:51.23999885469675
2025-09-29 06:41:45	energyRemaining:51.259998854249716

Bre77, 2025-09-29:
Isn't this easily mitigated by setting `minimum_delta` to `0.1`?

I use a `minimum_delta` on every single float/real fleet telemetry field: https://api.teslemetry.com/fields.json

Urkman, 2025-09-30:
From the changeling:

> Location fields now support minimum_delta. Changes in distance are measured in meters.